enwi/RNGBridgeDoc #4: Add Homeassistant MQTT auto discovery

enwi, 2022-02-25:
**Is your feature request related to a problem? Please describe.**
It would be great if RNGBridge would integrate into Homeassistant directly.

**Describe the solution you'd like**
Use MQTT auto discovery so that Homeassistant can find and display charge controller data automatically.

**Describe alternatives you've considered**
This could also be achieved through node-red, but requires more work and knowledge.

JochenKlenk, 2022-05-09:
Do someone have a template config howto add the json mqtt output to Homeassitant?

JochenKlenk, 2022-05-09:
HA only

enwi, 2022-05-09:
Then for the time being I will create the necessary templates and post them here so anyone can add them to their home assistant config.

enwi, 2022-06-27:
@JochenKlenk Sorry I have been busy with moving to a new apartment, but did you already create your own templates or do you still need help with this?

enwi, 2022-07-16:
Related to #7

JochenKlenk, 2022-07-21:
Hi,

sorry for the late answer. Would be great to have the templates or if you implment it into Autodiscovery then I can wait for it.

Thanks

enwi, 2022-07-21:
@JochenKlenk Sneek peak
[image]

enwi, 2022-07-22:
Smol update 🚀 
[image]

JochenKlenk, 2022-07-25:
Can't wait to get the new 2.6.10 :-)

enwi, 2022-07-25:
@JochenKlenk should not take that long until release, but I think I will also include #11 in the same release.

Also do you think it would be good to be able to change the Homeassistant publish path? The default is `homeassistant`, but technically you can change it in the HA integration settings

JochenKlenk, 2022-07-25:
Would be good if it is possible to change it.

enwi, 2022-07-30:
Feature added with [V2.7.0](https://github.com/enwi/RNGBridgeDoc/releases/tag/2.7.0)

closed 2022-07-30 by enwi