enwi/RNGBridgeDoc #5: Out of Stock

trentjw, 2022-03-12:
Looks like you lost interest in building these units

any Chance on sharing the Gerber Files for the board and schematics on it so it can still be used thanks

enwi, 2022-03-13:
@trentjw I did not loose interest, but I am working on a new hardware version (V2) with an improved and optimized user interface. The new version will have a few new features:
- Controlling load output through webinterface and rest API
- Over the air (OTA) software updates
- Ability to control the load output automatically according to state of charge with customizable set points
- Three open collector outputs that can also have set points according to state of charge or other data coming from the charge controller

enwi, 2022-03-13:
I hope that the new version will be available by the end of this month ☺️

trentjw, 2022-03-13:
okay sounds good

trentjw, 2022-03-14:
since you are redoing this maybe consider configurating the layout to work inside a Raspberry pi 4 case

enwi, 2022-03-14:
PCB's are already ordered, but I think it should at least fit inside. I might even make a 3D-printed housing.

closed 2022-03-14 by enwi

enwi, 2022-03-20:
@trentjw They are back in stock now 😉