enwi/RNGBridgeDoc #7: remote turn on of outputs 1 2 and 3

calmyabadselfdown, 2022-06-23:
I would like to be able to turn lights on based on the time of day and off to not over discharge the battery.
The status of the battery is already catered for so the off is covered. 

A MQTT option would work as that would allow integration with home automation options like Homeassistant 
( other home automation options are available :D  ) 
These are capable of publishing a turn on request based on various options like is it dark at your location.

Alternative option is some type of http request that again could be generated from a home automation platform

enwi, 2022-06-27:
Related to #8 

- [ ] Add MQTT interface for controlling outputs 1-3 and load output

enwi, 2022-07-30:
Feature added with [V2.7.0](https://github.com/enwi/RNGBridgeDoc/releases/tag/2.7.0)

closed 2022-07-30 by enwi