enwi/RNGBridgeDoc #8: Switch additional outputs 1,2,3 using webgui and rest-api

Doggy77, 2022-06-26:
I would like to be enabled to switch the additional outputs 1 to 3 using a Web gui switch and using a Rest API command just like the main load switch of the Renogy Controller. 

This would make it much more useful for automating processes in my solar powered garden shed, e.g. switching the water supply for my flowers. 

The switching only using the state of charge is too unflexible for my ideas.

enwi, 2022-06-27:
- [ ] Add output panel with switches and current state for outputs 1-3
- [ ] Add rest API to control outputs 1-3

enwi, 2022-07-16:
Output panel added with [release 2.5.0](https://github.com/enwi/RNGBridgeDoc/releases/tag/2.5.0)

closed 2022-07-16 by enwi