enwi/RNGBridgeDoc #11: Enhancement: read charge controller data using REST

Doggy77, 2022-07-24:
It would be cool to use the REST API for reading the charge controller parameters according to the MQTT response.

In my use case the MQTT data is sent using a 4G Router from my garden shed to my server. A second raspberry oder microcontroller can act according to the charge controller states, e.g. switch something off when an errror appears or switch something on when the sun goes down (panel voltage drops to zero). This can be done by subscribing to the MQTT, but as soon as the 4G would fail there is no possibility to get the controller data locally on the LAN.

By using GET requests for the REST API and reading the states directly from the RNGBridge an offline/failsafe operation would be possible.

enwi, 2022-07-24:
So you mean that you set up RNGBridge in access point mode and then have another ESP8266 connect to said AP and poll the REST api?

Doggy77, 2022-07-24:
No. At my garden I have an access point Router with 4g Internet connection and a Raspberry pi with wireguard VPN für tunneling all data traffic to my home server , which is also the MQTT server that collects the MQTT data. The Raspberry pi in the garden also controls shades, collects environment data and controls the watering pump for the produce.
The watering pump shall switch off or not switch on via GPIO as long as e.g. the battery is too low. For this I would be able to read the rngbridge's data weithin the local LAN without the need to rely on the VPN and the 4g connection in order to read data from my MQTT home server. Just by Rest API GET calls directly to the WiFi connected rngbridge.

enwi, 2022-07-24:
Should be fairly easy to implement :)

Doggy77, 2022-07-24:
In a brief: Getting the same json as from MQTT by http get. IP/api/status as according to IP/api/control for setting the output maybe.

enwi, 2022-07-30:
I have been thinking about this feature, but since `RNGBridge` already sends it's data using an [EventSource](https://developer.mozilla.org/de/docs/Web/API/EventSource) wouldn't it be easier to just connect to that? I mean it sends a status every second, so you would have to handle all that data. Or do you think polling a REST API is still better suited?

enwi, 2022-07-30:
REST api for getting state added with [V2.7.0](https://github.com/enwi/RNGBridgeDoc/releases/tag/2.7.0)

closed 2022-07-30 by enwi