enwi/RNGBridgeDoc #12: WiFi Signal Strength Indication

Matt86C, 2022-08-03:
**Is your feature request related to a problem? Please describe.**
I have the RNGBridge running in a garden shed which has patchy WiFi. Having some feedback on the WiFI signal strength via MQTT would help a lot with positioning the unit.

**Describe the solution you'd like**
MQTT topic including WiFi signal strength. Ideally also an indication of this on the web UI.

**Describe alternatives you've considered**
The alternative is blindly adjusting RNGBridge location and looking to see if WiFI connection is lost in Home Assistant.

**Additional context**
I use an ESP for controlling my boiler using the following project. This integrates with home assistant really well and might be a useful reference case. https://github.com/emsesp/EMS-ESP

enwi, 2022-08-04:
Sure, this can be added fairly easy

enwi, 2022-08-04:
New version coming soon:
[image]

enwi, 2022-08-04:
Added with [V2.7.2](https://github.com/enwi/RNGBridgeDoc/releases/tag/2.7.2)

closed 2022-08-04 by enwi

Matt86C, 2022-08-04:
Amazing, thanks for the quick turn around!

enwi, 2022-08-04:
You are welcome 🙂