enwi/RNGBridgeDoc #13: REST API Get - Missing Parameters regarding power

Doggy77, 2022-08-07:
Version used: 2.7.3

When sending a GET Request to http://<RNGIP>/api/state the json answer is missing the "l":{"po"} and the "p":{"po"} parameters:

`{"mqttsta":"Connected","pvosta":"Disabled","b":{"ch":100,"vo":13.69999981,"cu":1.080000043,"te":26},"l":{"vo":0,"cu":0},"p":{"vo":19,"cu":0.779999971},"c":{"st":5,"er":0,"te":30},"o":{"l":false,"o1":false,"o2":false,"o3":false},"up":300,"he":38048,"rssi":-35}`

The load is switched off therefore the power is "0", but the panel delivers some Watts (see screenshot).

https://github.com/enwi/RNGBridgeDoc/blob/main/rest.md#data-values mentions these parameters, but they already lack in the paragraph above https://github.com/enwi/RNGBridgeDoc/blob/main/rest.md#data-structure

[image]

enwi, 2022-08-07:
Good call, these two power values are missing here, because they don't need to be sent to the GUI, because you can just multiply the voltage and current to get the power. I might just remove all power values instead, which results in less data needing to be sent/received/processed.

Doggy77, 2022-08-08:
If these values are not sent by REST API intentionally/deliberately the documentation needs to be corrected. 

Personally I would prefer to see the Watts in GUI because it is more intuitively to see and saves me from bad calculating in mind e.g. to see if the panel is near its maximum. This also applies to the json answers in REST/MQTT. 

The advantage of saving data on the other side would be bigger if the values would be rounded to 1 or 2 decimals even with both power values included.

enwi, 2022-08-08:
> If these values are not sent by REST API intentionally/deliberately the documentation needs to be corrected.
Yes

> Personally I would prefer to see the Watts in GUI because it is more intuitively to see and saves me from bad calculating in mind e.g. to see if the panel is near its maximum. This also applies to the json answers in REST/MQTT.
The watts in the GUI will still be displayed and are currently calculated by multiplying votlage with current. (`P=U*I`). In my opinion multiplying two values isn't that bad.

> The advantage of saving data on the other side would be bigger if the values would be rounded to 1 or 2 decimals even with both power values included.
That would also be an option.

Doggy77, 2022-08-09:
As long as it is displayed in GUI I don't care if it is calculated or read out from the RENOGY :-)

And I can live with it to calculate the values myself when using the REST GET.

Therefore only the documentation change remains open.

enwi, 2022-08-09:
Great, I will change everything with the next release, which is already in the making, featuring some improvements and changes.

enwi, 2022-10-03:
Fixed with [2.8.0](https://github.com/enwi/RNGBridgeDoc/releases/tag/2.8.0)

closed 2022-10-03 by enwi