enwi/RNGBridgeDoc #15: External switch to control Output 1, 2 and 3 physically

klu03rl, 2023-09-04:
**Is your feature request related to a problem? Please describe.**

For the wife approval factor, it would be great to be able to use the rngBridge with a physcial switch to control the outputs to a relay so that the app doesn't need to be used, which can be inconvienient to do all the time when a physical switch that triggers the output on / off would be easier for any end user.

**Describe the solution you'd like**

Be able to switch the outputs of the rng bridge on / off from a physical switch as well as through the web UI / MQTT etc.

**Describe alternatives you've considered**

Not tried any other options, potentiall could look at other mechanism to control relay, but then the status would not align with the status of the RNGbridge / home assitant etc which could impact automations.

A clear and concise description of any alternative solutions or features you've considered.

not tried.

**Additional context**

N/A

klu03rl, 2023-09-04:
Presume this would likely require a hardware revision

enwi, 2023-09-04:
If you are using HomeAssistant, you could add a wireless switch or button to it to switch/toggle any of the RNGBridge outputs

enwi, 2024-08-18:
@klu03rl Any updates on this? Since the source code is open source now, you could use some of the free GPIOs to connect three buttons and change the code.

closed 2024-08-18 by enwi