enwi/RNGBridgeDoc #17: Battery Temperature below 0°C is schown as -127°C

Doggy77, 2024-01-08:
**Describe the bug**
Battery Temperature below 0°C is shown as -127°C, in API and in Web

`{"mqttsta":"Disabled","pvosta":"Disabled","o":{"o1":false,"o2":false,"o3":false,"l":true},"up":645037,"he":37992,"rssi":-82,"b":{"ch":95,"vo":13.10000038,"cu":0.620000005,"te":-127,"ge":38,"co":76},"l":{"vo":13.10000038,"cu":0.620000005},"p":{"vo":31.5,"cu":0.25},"c":{"st":2,"er":0,"te":5}}`

[image]

(ATM we have ca. -0,6°C where the battery is located)

Original sensor is used. Controller temperature did not fall below 0, but maybe here is the same issue.

Low priority, if this is sent by RNG itself it doesn't matter, I have another (Tasmota) sensor and can live without this.

enwi, 2024-01-09:
Thanks for the bug report, I will take a look :)

Doggy77, 2024-01-09:
Addendum: It seems as if -127 equals -0°C, the lower the temp, the higher the value climbs: approx. real -6°C come out as -122. 
I conclude: It is not stuck at -127.
Thanks for your work!

enwi, 2024-01-22:
Ok I think I found the issue. I need to strip off the MSB for it to read correctly.
Taken from the datasheet: `(b7: symbol bit；b0-b6 temperature value)`

enwi, 2024-01-22:
I built a new binary, can you see if it fixed the issue? (I hope it is still cold enough to test 🤣 )
[binary.zip](https://github.com/enwi/RNGBridgeDoc/files/14015373/artifact.zip)

Doggy77, 2024-01-22:
I downloaded your ZIP and flashed it. It shows 2.12.0 and _Update available_ (which I didn't install yet): [image]

Unfortunately it isn't cold enough anymore:
[image] but I will have an eye on it.

Doggy77, 2024-01-26:
I can confirm that the issue is solved, both in WebUI and also in API-REST response
[image]
Thank you!
P.S.: It shows "New version available" and if I click this it reverts to 2.11?

enwi, 2024-01-26:
Great that the issue is resolved, I will release the new version then.

The "new version available" message is another bug, it will be shown if the latest version on GitHub does not match the one on the device 😅

closed 2024-08-18 by enwi