enwi/RNGBridgeDoc #18: Question on troubleshooting power

lma2022, 2024-01-28:
I'm having trouble getting the rngBridge to power up.  My Renogy charge controller is pushing out 14.5V on the correct pins.  With the buck converter plugged in, I don't see any activity on the rngBridge.
I tried removing the buck converter and measure GND, 5V, 5V on the 3 pins where the buck converter plugs into (even though they are labelled 15V,GND,3V3 respectively.

Any suggestions on how to troubleshoot this?

Could the buck converter be faulty?  Or maybe some of my soldering somewhere ;(

thanks

enwi, 2024-01-28:
Did I understand you correctly that you measured GND for the 15V pin, 5V for the GND pin and 5V for the 5V pin while the buck converter is removed?

enwi, 2024-01-28:
Also can you tell me the model of your charge controller?

enwi, 2024-01-28:
> Could the buck converter be faulty?

Generally I test all components individually as well as connecting them to a spare charge controller before shipping, so everything should work as expected. Though this does not mean that things can break during shipping or due to ESD damage.

lma2022, 2024-01-28:
Hi, it's a Renogy Rover 40A charge controller.
Yes, with the buck removed, and the RJ12 cable connected to the charge controller, I measured on the pins:
pin #1 labelled 15V = GND
pin #2 labelled GND = 5V (compared to pin #1)
pin #3 labelled 3V3 = 5V (compared to pin #1)

That made me very confused!! :)

enwi, 2024-01-29:
Did you power the ESP8266 with a USB cable while measuring?

enwi, 2024-01-29:
Before you do that, what RJ12 cable are you using? A straight or reverse one?
[image]

lma2022, 2024-01-29:
You my friend, are brilliant!
:1st_place_medal: 
Yes it is indeed crossed-over...dammit I should have noticed that !

I will amend that and come back to you (hopefully) with good results!

(that makes sense with the 5V now...coming from the signal wires....)

enwi, 2024-01-29:
I hope nothing was damaged 😅

enwi, 2024-02-04:
Did you solve the issue @lma2022 ?

lma2022, 2024-02-05:
> Did you solve the issue @lma2022 ?

Sorry, haven't had a chance to try yet....planning to do so this coming weekend...

lma2022, 2024-02-09:
Ok, I do have an update....I used a straight through cable and it powered up okay.  I see the AP and joined the RNG WLAN as well!  But that's where the good news ends...
I couldn't open the web page, and tried a ping tool to 192.168.4.1 but there was no response.
Both the orange LED and blue LED would flash each second in unison.

Could this mean I've burnt out something?

enwi, 2024-02-09:
@lma2022 If you see a short orange flash followed by a longer orange flash and a blue flash then the device should be operational

enwi, 2024-02-09:
During initialization (after a reset or power cycle) the orange led should be fully lit

lma2022, 2024-02-11:
You were right :)  I tried again, but this time with my laptop rather than an Android phone - and the portal screen displayed for me...and all is well!

Thank you for your guidance @enwi

enwi, 2024-02-11:
You are welcome, I think product support should always be like this 😅

closed 2024-02-11 by enwi